xperiments-in/xtouch #12: Boot hang - MQTT Error when IP of printer changes

hewfish, 2023-10-10:
If your bambu labs printer is on a dynamic IP on wireless, and changes IP, the MQTT connection will fail on boot as the IP programmed into the configuration file stored on the SD card is no longer available.

**To Reproduce**
Turn your printer off and on again after 1 day with no IP reservation on a network with high change. printer gets new IP and xtouch will no longer boot.

**Expected behavior**
If the printer is unavailable, go to the main screen and ask to either connect to a new printer or re-scan for the printer. Or have the stack scan the IP range for a printer that can be connected to with the access code programmed into the configuration already.

**XTOUCH (please complete the following information):**
 - FW: latest

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version: latest

xperiments, 2023-10-10:
Can you come to the discord channel. I am finishing a fix. I will like you to test. 

https://discord.gg/KrwfjHsY

Thanks

xperiments, 2023-10-14:
Fixed in 0.9.14

closed 2023-10-14 by xperiments