xperiments-in/xtouch #14: Manually extrude filament - Filament at 170C? - bug or feature

en-x, 2023-10-12:
Not sure if this is a bug or how the feature is supposed to work. I wanted to report it since I find it kind of odd how it works.
After I load new filament using the P1Touch, I always manually extrude additional filament to be sure my first print is clean. When I press the extrude button on the P1Touch, I get a popup message asking if the Nozzle at 170C: No / Yes. The Nozzle is always at this temp since I just loaded the new filament. I click Yes and it extrudes a small amount of filament. Then I have to click the down arrow again to extrude and am prompted again for the filament temp.

Is there a way to read the filament temp and only prompt if the temp is not at 170C or only ask once to allow me to extrude continuously while holding down the button or pressing it multiple times.

To reproduce: load new filament or set the filament temp to 250C. Select the manually extrude down arrow and you should receive the prompt.

P1Touch version: 0.8.17
Printer: P1S
Firmware: 01.04.01.00
No AMS

Great job with this software and thanks again for letting be part of the beta test.

[image]
[image]
[image]

xperiments, 2023-10-17:
Ok I will take care of this.
What we will do. I check if nozzle tem is < 170. If so the dialog appears. If not you can click as much as you what...
Makes sense?

en-x, 2023-10-17:
Sounds great! Thank you!

commit 99f2b20b (2023-10-17): Fixes #14
closed 2023-10-17 by xperiments