xperiments-in/xtouch #18: Online installation & temp sensor

WimVDK, 2023-10-15:
On the verge of buying the components for this (as i'm days away from ordering my p1s with AMS). This is more 2 questions rather than an issue, hence the blank "issue". 

Q1: I was wondering if there was a way to do an offline installer, or what the reason is for online one. My main browser is Firefox (and i'd have to install chrome) which doesn't seem to support the online installer?

Q2: Does the temp sensor need any soldering? The instructions on full linking and installation are a bit lacking on that point?

Otherwise excellent "mod" and thanks for making it available to the 3D community for free!

xperiments, 2023-10-15:
Hello thanks for your comments

Q1: you can flash directly the OTA published file now at 0.9.14

http://xperiments.in/xtouch-bin/ota/xtouch.0.9.14.bin

Not included docs as mostly ppl will go the chrome path... and not all are tech...

Q2: there are 100th variations of the sensor, with shield , without, small, bigs, water proff.

So it will depend of the sensor y choice... but for the one I provided, you need a 1.25T 4P JST connector to dupont. 
BTW you mostly have this cable coming w the screen. 

I close the non issue. BTW if you need help come to discord

closed 2023-10-15 by xperiments

WimVDK, 2023-10-15:
would you mind linking such a connector on aliexpress (where you linked the screens), so i can order everything in 1 go :). Add the discord too if you don't mind :)

xperiments, 2023-10-15:
I don't have a link for the connector as mine come with my screen as commented( The linked one's from ali come with it when ordered)
For the length of the able it will depend a lot of your install , but having this one you an easy to extend it
I am sorry to don't provide more help...