xperiments-in/xtouch #25: Connection to printer not possible, printer not found...

kohlsalem, 2023-10-19:
**Describe the bug**
Setup & Flashing worked fine as described in the readme.md.; Connection to Wifi worked, however, the devices earches for hours without finding my p1p.

**Expected behavior**

Would be happy to connect to the printer :-)

**Screenshots**
[image]

**XTOUCH (please complete the following information):**
 - Setup with latest firmware (day before yesterday), can not confirm any number since it is searching for printers after starup

**Printers (please complete the following information):**
 - Device: P1P
 - FW Version 01.04.01.00 (Latest VErsion)
-  AMS units:nope

**Additional context**
- AVM Fritz Box with Several W-LAn Bridges
- Normal 192.168.178.x Adress Range
- no VLAN etc. PP
- All parties are members of the notmal WLAN.
- While xTouch can not find the printer, i can see printer and cam in BambuStudio
- Printer not in "Lan Mode" but in normal Cloud mode
- Good WLAN Connection
- no known general issues connecting between device

It looks to be like i have a pretty stupid oversight here, any idea?

Thanks
Best
Michael

xperiments, 2023-10-19:
Not sure, as wifi is connecting. Here is searching in ssd port for printers. Mostly related to your wifi setup...
To discuss this kind of topic better come to discord.

Thanks!!!

kohlsalem, 2023-10-19:
Just found the logs.. Looks... not normal.
```
[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.178.64
[XTouch][SD] Reading JSON file: /xtouch/printer-ips.json
[XTouch][SD] Error Reading JSON File: /xtouch/printer-ips.json
[XTouch][SD] Reading JSON file: /xtouch/printer-pair.json
[XTouch][SD] Error Reading JSON File: /xtouch/printer-pair.json
[XTouch][SD] Reading JSON file: /xtouch/printer-ips.json
[ 15137][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x0d
[ 15138][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
[ 15441][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 15744][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[XTouch][SD] Error Reading JSON File: /xtouch/printer-ips.json
[XTouch][SD] Writting JSON file: /xtouch/printer-ips.json
[ 16059][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 16362][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 16666][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 16969][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 16969][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sd/xtouch/printer-ips.json) failed
[XTouch][SD] Failed to write json file: /xtouch/printer-ips.json
[XTouch][SD] Reading JSON file: /xtouch/printer.json
[ 17285][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 17588][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[XTouch][SD] Error Reading JSON File: /xtouch/printer.json
[XTouch][SD] Writting JSON file: /xtouch/printer.json
[ 17902][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 18206][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 18509][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 18812][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 18812][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sd/xtouch/printer.json) failed
[XTouch][SD] Failed to write json file: /xtouch/printer.json
[XTouch][SD] Reading JSON file: /xtouch/printer-ips.json
[ 25175][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 25479][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[XTouch][SD] Error Reading JSON File: /xtouch/printer-ips.json
[XTouch][SD] Writting JSON file: /xtouch/printer-ips.json
[ 25794][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 26097][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 26401][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 26705][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 26705][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sd/xtouch/printer-ips.json) failed
[XTouch][SD] Failed to write json file: /xtouch/printer-ips.json
[XTouch][SD] Reading JSON file: /xtouch/printer.json
[ 27021][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 27324][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[XTouch][SD] Error Reading JSON File: /xtouch/printer.json
[XTouch][SD] Writting JSON file: /xtouch/printer.json
[ 27638][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 27941][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 28244][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 28547][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 28547][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sd/xtouch/printer.json) failed
[XTouch][SD] Failed to write json file: /xtouch/printer.json
```

kohlsalem, 2023-10-19:
SD Card reformatted - no help.
New SD Card - works like a charm.
Case closed :-)

closed 2023-10-19 by kohlsalem

kohlsalem, 2023-10-19:
Beside my issues with one card: there seems to be as well an upper sd card size limit of 16.  Another card with 32 did not work.

I assume it was really size and no other compatibility issue.  May be it would be worth mentioning in the read.me...