xperiments-in/xtouch #30: xtouch keeps prompting to select the printer

mike1808, 2023-10-22:
**Describe the bug**
From time to time when I check on xtouch I see the printer selection prompt.

**To Reproduce**
N/A
Unfortunately, I'm unable to provide repro steps. 

**Additional context**

xperiments, 2023-10-22:
This is mostly related to wifi reconnections. And mostly to DHCP changes. Please check your wifi router settings.

Closing ATM. Reopen if needed, but please try to confirm before opening a bug in discord channel

closed 2023-10-22 by xperiments

mike1808, 2023-10-25:
I have checked all the logs and the IP addresses of xtouch and the printer doesn't change. I also fixed the address and it still has same issues.
I'd like to keep the discussion in GitHub so if other have similar issues they can find it through search engines.

xperiments, 2023-10-25:
Have you checked with the latest version of today?