xperiments-in/xtouch #32: Expose chamber temperature to MQTT

gillesguillemin, 2023-10-24:
My understanding is that xtouch uses MQTT under the bonnet to access and control the printer.

Would it be possible to expose the chamber's temperature (when using a DS18B20 sensor) to MQTT so it could be accessed by Home Assistant, so it could be automated?

This is all assuming this is not already possible somehow (in which case I'm curious to know how).

Alternatively, could it be possible to implement some sort of chamber temperature control by tying the chamber fan to the sensor's temperature?

Thanks (and great work btw)!

xperiments, 2023-10-24:
That's have been already tested. It works but not all... Because actually the printer is sending back a value of 5. Need to check this in more detail...

xperiments, 2023-12-04:
Yes, but the thing is p1x series report a 5deg of chamber temp, as they don't have it. So sending the correct values to the BBS will work but time to time values will read 5deg

xperiments, 2023-12-17:
I will close this ATM. As is a little hack to put it to work. Also values are not consistent

closed 2023-12-17 by xperiments