xperiments-in/xtouch #34: Error

Juststartingout1, 2023-10-26:
this is the error I keep getting. ive tried different cords not sure if any of them are data cords. I honestly don't know how to check.

XTouch][SCREEN] Setup
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f54fe  PS      : 0x00060230  A0      : 0x800d26fb  A1      : 0x3ffb2150  
A2      : 0x00000000  A3      : 0x51c00000  A4      : 0x000000ef  A5      : 0x00000000  
A6      : 0x00000055  A7      : 0x00000000  A8      : 0x800f54fe  A9      : 0x3ffb2130  
A10     : 0x0000024d  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x0000024d  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000004  LBEG    : 0x40084815  LEND    : 0x4008481d  LCOUNT  : 0x00000027  

Backtrace: 0x400f54fb:0x3ffb2150 0x400d26f8:0x3ffb2170 0x400f34f6:0x3ffb21a0 0x400e6ac3:0x3ffb21c0 0x400f5480:0x3ffb2210 0x400d2c7f:0x3ffb2240 0x400db15a:0x3ffb2260 0x40113672:0x3ffb2290

ELF file SHA256: a330626b78874a13

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[XTouch][SCREEN] Setup
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f54fe  PS      : 0x00060230  A0      : 0x800d26fb  A1      : 0x3ffb2150  
A2      : 0x00000000  A3      : 0x51400000  A4      : 0x000000ef  A5      : 0x00000000  
A6      : 0x00000055  A7      : 0x00000000  A8      : 0x800f54fe  A9      : 0x3ffb2130  
A10     : 0x0000024d  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x0000024d  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000004  LBEG    : 0x40084815  LEND    : 0x4008481d  LCOUNT  : 0x00000027  

Backtrace: 0x400f54fb:0x3ffb2150 0x400d26f8:0x3ffb2170 0x400f34f6:0x3ffb21a0 0x400e6ac3:0x3ffb21c0 0x400f5480:0x3ffb2210 0x400d2c7f:0x3ffb2240 0x400db15a:0x3ffb2260 0x40113672:0x3ffb2290

ELF file SHA256: a330626b78874a13

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[XTouch][SCREEN] Setup
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f54fe  PS      : 0x00060230  A0      : 0x800d26fb  A1      : 0x3ffb2150  
A2      : 0x00000000  A3      : 0x51800000  A4      : 0x000000ef  A5      : 0x00000000  
A6      : 0x00000055  A7      : 0x00000000  A8      : 0x800f54fe  A9      : 0x3ffb2130  
A10     : 0x0000024d  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x0000024d  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000004  LBEG    : 0x40084815  LEND    : 0x4008481d  LCOUNT  : 0x00000027  

Backtrace: 0x400f54fb:0x3ffb2150 0x400d26f8:0x3ffb2170 0x400f34f6:0x3ffb21a0 0x400e6ac3:0x3ffb21c0 0x400f5480:0x3ffb2210 0x400d2c7f:0x3ffb2240 0x400db15a:0x3ffb2260 0x40113672:0x3ffb2290

ELF file SHA256: a330626b78874a13

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[XTouch][SCREEN] Setup
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f54fe  PS      : 0x00060230  A0      : 0x800d26fb  A1      : 0x3ffb2150  
A2      : 0x00000000  A3      : 0x51800000  A4      : 0x000000ef  A5      : 0x00000000  
A6      : 0x00000055  A7      : 0x00000000  A8      : 0x800f54fe  A9      : 0x3ffb2130  
A10     : 0x0000024d  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x0000024d  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000004  LBEG    : 0x40084815  LEND    : 0x4008481d  LCOUNT  : 0x00000027  

Backtrace: 0x400f54fb:0x3ffb2150 0x400d26f8:0x3ffb2170 0x400f34f6:0x3ffb21a0 0x400e6ac3:0x3ffb21c0 0x400f5480:0x3ffb2210 0x400d2c7f:0x3ffb2240 0x400db15a:0x3ffb2260 0x40113672:0x3ffb2290

ELF file SHA256: a330626b78874a13

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[XTouch][SCREEN] Setup
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f54fe  PS      : 0x00060230  A0      : 0x800d26fb  A1      : 0x3ffb2150  
A2      : 0x00000000  A3      : 0x52c00000  A4      : 0x000000ef  A5      : 0x00000000  
A6      : 0x00000055  A7      : 0x00000000  A8      : 0x800f54fe  A9      : 0x3ffb2130  
A10     : 0x0000024d  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x0000024d  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000004  LBEG    : 0x40084815  LEND    : 0x4008481d  LCOUNT  : 0x00000027  

Backtrace: 0x400f54fb:0x3ffb2150 0x400d26f8:0x3ffb2170 0x400f34f6:0x3ffb21a0 0x400e6ac3:0x3ffb21c0 0x400f5480:0x3ffb2210 0x400d2c7f:0x3ffb2240 0x400db15a:0x3ffb2260 0x40113672:0x3ffb229

xperiments, 2023-10-26:
This error is mostly because you are not using the correct screen board.
Where do you get it?
Can you share some image?
Please enter discord before opening a issue...
There we can help more than here.

I close ATM. Reopen if needed

closed 2023-10-26 by xperiments

Juststartingout1, 2023-10-26:
Aliexpress
https://www.aliexpress.com/item/3256804913272047.html?spm=a2g0o.order_list.order_list_main.11.14ec1802RG7JuL

This is the wrong board. I changed the size to see it better.
sorry i'm old and just trying to make the printer easier to use
thank you for your help

xperiments, 2023-10-26:
It's not the good one. I am sorry. Size matters. You need to change for
2.8. I am working in the 5 inch but no ETA