xperiments-in/xtouch #35: Screen doesn't show all "buttons" in event of a problem

bpsmicro, 2023-10-26:
For the first time, I had a failure to unload a filament via AMS after a print.

xTouch correctly showed me the message, but where it popped up the message on what to do, there's *supposed* to be a "Confirm" and "Retry" button (idea is you try to manually free up the spool and then click Retry).
Only the "Confirm" button was shown. I assume "Retry" would be underneath that, but off the screen. I thought maybe I could scroll, but that seemed not to do anything.

I ended up grabbing my phone and using Bambu Studio to fix the printer, once I figured out it was trying to get me to push a button I couldn't see. :-)

Didn't get screenshots, but I'm going to see if the problem (with the unload) happens again tomorrow and if so I'll get more specifics.

gillesguillemin, 2023-10-26:
It behaves in the exact same way as Bambu Studio: you just need to hit confirm after you fixed the issue, and then hit the "play" button to resume the print.
It could be a tad clearer, I agree, but that's the expected behaviour as defined by Bambu Lab as far as I can tell.

xperiments, 2023-10-26:
Can I then close?

Thx!!

bpsmicro, 2023-10-26:
Hmm, okay, thanks for that. I was using my phone with Bambu Studio, which showed additional buttons. I failed to look at the PC screen running Studio.
I'll pay better attention next time (which could be tomorrow, or never, who knows).
What was confusing was I pulled the filament out and hit Confirm, but nothing happened and the AMS light was flashing red for that spool. But when I looked at the phone, there was the extra button to retry (after fixing), and that "reset" the AMS.

I'll play with this more tomorrow and verify.

xperiments, 2023-10-26:
As far as I know, they are changing a lot the messages for the errors.
xtouch get the updated messages on every release, but somes can be
different..

In the future I will add a build robot to auto publish patch versions of
xtouch and generate a new OTA automatically...

bpsmicro, 2023-10-27:
Okay, I had the retraction problem again today, and spent more time looking at the message in Studio, Handy and on the xTouch. And indeed. the wording of the message implies there should be a "Retry" button right there, but there isn't. Just "Confirm", and then you can go look for the Retry in the AMS control settings.
Since the xTouch doesn't yet **have** AMS control settings, the message was just slightly confusing. But that's not the xTouch's fault.

I'm closing this. Thanks for the clarification.

closed 2023-10-27 by bpsmicro

xperiments, 2023-10-27:
Thank to you!!