xperiments-in/xtouch #36: Does not resume print properly

s3anami, 2023-10-27:
I**Describe the bug**
I selected resume print from the screen after replacing my filament, the printer proceeded to crash itself into all the prints bending the hot end and then proceeded to smash against the back left corner of the printer repeatedly attempting to home or something.

**To Reproduce**
Steps to reproduce the behavior:
Load Filament button
Confirm
Click Resume Print

**Expected behavior**
To resume printing properly

**Screenshots**
If applicable, add screenshots to help explain your problem.

**XTOUCH (please complete the following information):**
 - FW: Latest just updated

**Printers (please complete the following information):**
 - Device: P1P upgraded to P1S
 - FW Version Latest
-  AMS units: 0

kenkamm, 2023-11-21:
Is this still an issue? Scary...

xperiments, 2023-12-16:
@s3anami Can you check with the latest firmware? Thanks

xperiments, 2023-12-17:
I will close as I think is a dupplicate of #47 that is already solved. Feel free to reopen if needed

closed 2023-12-17 by xperiments