xperiments-in/xtouch #38: Offline Install Option

codewizard1975, 2023-10-31:
I've been searching and cannot find the FIRMWARE.BIN file as described anywhere to download.  A lot of people cannot use the online installer as they're using unsupported browsers, so why make it so hard to use your project?

xperiments, 2023-10-31:
Mostly because next install version will require web installer
Also for a non tech ppl browser install is way easy

Meanwhile you can download the bin from xtouch-bin , rename it to firmware
and put it on the screen

I am not home this week but if y enter discord, not too far in time I
explained there how to do

codewizard1975, 2023-10-31:
Thank you for providing info on how to get it.  

I'm actually running on Linux and use Firefox so can't do the online install option.  I know from the past that browsers like Brave and Opera are not supported but used by many people.

I'd recommend adding an actual link in the README.md to the xtouch-bin repository for those who are techy.

closed 2023-10-31 by codewizard1975

xperiments, 2023-10-31:
I understand but as commented I need the webinstall as I modified the
current WiFi setup and is done via serial. Anyway I will add an install
path for techies (if actually I can escape the browser thing)