xperiments-in/xtouch #39: During load funtion, retry is not shown

aidzer0, 2023-10-31:
**Describe the bug**

Hello! Firstly, amazing work on this project, it is amazing to use!

In BambuStudio there are 3 options when trying loading new filament: "Confirm", "Done", "Retry", however the xtouch does not have the "Retry" option, it only has the "Confirm", "Done" options after the first load cycle. The retry is used if the extruder is still not spitting the filament out on the first try. 

 I noticed there was another issue #35 similar to this opened and closed. But the retry option is still in bambu studio and the handy app, if possible, of course, it would be great to have that button added.

**To Reproduce**
Steps to reproduce the behavior:
1. Just try to load new filament and the option is not there once it cycles through the first batch of jobs of the load function.

**Expected behavior**
A clear and concise description of what you expected to happen.
It would be nice to remove the "Confirm" option with the "Retry", if possible. I realize that there might not be enough screen real-estate to add a third option. 

**Screenshots**
If applicable, add screenshots to help explain your problem.
I did not get any this time, I am about to run out of filament so I will upload some as soon as I do.

**XTOUCH (please complete the following information):**
 - FW: 0.9.24

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version: 01.04.02.00
-  AMS units: None

**Additional context**
Thank you for all your hard work on this project, it is absolutely amazing!!! And make the P1 series even better!

aidzer0, 2023-11-12:
Here are the screenshots of Bambu Studio prompts and xTouch Prompts. Its not a huge issue by any stretch, this is one of those "comfort" optimizations, if there is time to look at. thank you!!

I have been using the xTouch for about 2 weeks now and it absolutely improves the P1S usability.

[image]
[image]

xperiments, 2023-11-12:
Yes, is "normal" because before lattest BBS there only 2 options...

I will fix this

aidzer0, 2023-11-12:
You are a legend, thank you!!!!

xperiments, 2023-12-13:
duplicate of #47

closed 2023-12-13 by xperiments