xperiments-in/xtouch #41: Crash on 4GB memory card

Error-106, 2023-11-05:
[**Describe]the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Insert 4GB memory card. (Kingston, written to wifi.json)
2. After calibrating the screen, enter the search interface and try binding.
3. Enter the 8-digit binding code.
4. Automatically restart and try binding again. (The log has been attached at the end)

**Expected behavior**
Enter the main interface

**Screenshots**
If applicable, add screenshots to help explain your problem.

**XTOUCH (please complete the following information):**
 - FW: 0.9.24

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version: 01.04.02.00
-  AMS units: None
**Additional context**
This bug seems to only happen on this 4GB memory card (other memory cards are fine)
[esp-web-tools-logs.txt](https://github.com/xperiments-in/xtouch/files/13259959/esp-web-tools-logs.txt)

xperiments, 2023-11-05:
Not all sd works fine, try another branch...

closed 2023-11-05 by xperiments

Error-106, 2023-11-05:
No, it was my mistake, this card is fake,sorry