xperiments-in/xtouch #49: Display current status / step of "Load Filament" process on xTouch display

brandonkish, 2023-11-06:
## Feature Description

1. Indicate the current status / step that the "Load Filament" process is at. i.e. "Heat The Nozzle", "Confirm filament has been extruded", "Purging old filament" 

## Justification

When attempting to load a new filament, there is no indication that the nozzle is heating, the "Load New Filament" window disappears and it looks as nothing happened. This can only be confirmed if you know to go back to the home screen to view the temperature of the nozzle. This should be clearer to the user, and displayed directly on the filament screen as some kind of status indicator. This is how the firmware on the Bambu P1S unit works. There is status text that indicates the current step in the process right on the load filament page. 
[image]
[image]
[image]
[image]

## Suggested Implementation (if you have one)

The current load status should be indicated on the filament page page, along with the current nozzle temperature.

## Screenshots or Examples (if applicable)

[image]
[image]
[image]
[image]

fdebeer, 2023-11-20:
2nd this

kenkamm, 2023-11-21:
This is definitely needed as well as some cleanup of the dialogs that come up when loading filament. For instance at one point it says to press retry if the filament has not loaded, but the only buttons presented are "Done" and "Confirm."

heelsmj, 2024-01-27:
Yes this would be great.