xperiments-in/xtouch #50: Pressing "Done" during load filament process does not turn off nozzle temperature.

brandonkish, 2023-11-06:
**Describe the bug**

1. Pressing "Done" button during the "Load" filament process does not reset the nozzle temperature to the cooling temperature.

**To Reproduce**
Steps to reproduce the behavior:
1. Power cycle Bambu Printer
2. Power cycle xTouch display
3. Go to "Filament screen" on xTouch.
4. Press "Yes" to confirm load process on xTouch.
[image]

5. Wait for nozzle to heat on Bambu Printer, and for Done / Confirm prompt to display on xTouch
[image]

4. Press the "Done" button on xTouch display.
4. Wait several minutes to confirm on the xTouch display and Bambu Printer that the nozzle temp is still at 250*C

**Expected behavior**

1. After pressing "Done" during the "Load" filament process, the nozzle temp should be set to a cool temp (~42*C) after a few seconds.

**Screenshots**
[image]

**XTOUCH (please complete the following information):**
 - FW: v0.9.24

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version: 01.04.02.00
-  AMS units: 0

**Additional context**

**NOTE** I believe some of the functionality for the "Confirm" and "Done" buttons are reversed. The "Confirm" button seems to cool the nozzle but, it shouldn't since it prompts again afterwards to confirm again. And the "Done" button is cooling the nozzle. I believe these are simply backwards.

kenkamm, 2023-11-22:
When loading, I press "Done" on mine and the filament will feed a bit as expected, but then the temp goes up to 260 and stays there.

xperiments, 2023-12-13:
duplicate of #47

closed 2023-12-13 by xperiments