xperiments-in/xtouch #60: Ability to disable preheat before extruder movement

cldesigns, 2023-12-01:
## Feature Description

I would love the ability to disable the requirement for the nozzle to be heated before the extruder gear can be moved in the settings.

## Justification

The filaments I print with almost always have to be cut not just unloaded (plus it's faster anyway) so I almost never preheat before moving the extruder. The default interface, app, and desktop app all let you move the extruder without heating.

## Suggested Implementation (if you have one)

add setting checkbox

## Screenshots or Examples (if applicable)

--------------------------------

Thank you for the hard work on this project! It's such an awesome upgrade for the P1S.

xperiments, 2023-12-23:
Makes sense for you if I remove the limitation for the Up button (so unloading) and remain the 170 warning for actually extruding?

cldesigns, 2023-12-23:
Totally. Yeah it makes sense to keep the extrusion warning on when feeding and an option to disable it for retracting. Also didn't know if you were aware that if you have the Bambu desktop app open when you dismiss a warning message on the touchscreen it infinitly pops up after any input until you dismiss the message via the desktop app. Sent from my Verizon, Samsung Galaxy smartphone

commit 9964a433 (2023-12-24): Closes #60
closed 2023-12-24 by xperiments