xperiments-in/xtouch #71: Control P1S chamber fan with Temp input

steve3j, 2023-12-16:
## Feature Description

Ability to power up/down chamber fan based on the temperature input from temp sensor.

## Justification

Currently, P1S chamber fan % is based on 

## Suggested Implementation (if you have one)

[Ideas on how the feature could be implemented]

## Screenshots or Examples (if applicable)

[Attach screenshots or examples if relevant]

rwijnhov, 2023-12-29:
I would like this too, came specialy here to add this as a feature request :)

m33ps, 2024-01-04:
This could be very helpful in more reliable prints as well as noise reduction.

heelsmj, 2024-01-27:
Yes this would be great.

DDen87, 2024-06-13:
This would be a big plus if possible!

nfnhayden, 2024-11-11:
or instead add a 3rd party heater to the chamber, and control it with the IO pin not used for monitoring the DS18B20 sensor, it could trigger a 3.3v relay that can use 110v to run the heater....

fjpg1970, 2025-05-14:
Has this option been implemented or is it not possible?

xperiments, 2025-05-14:
@nfnhayden You can't say I don't provide updartes or don't work on it
Do you know?
Have you even check the commit history?

[image]

Just this feature is not posible, we have already tested in the past, and the conclusion was it's not reliable...

closed 2025-05-14 by xperiments

nfnhayden, 2025-05-14:
I didn't mean you arn't maintaining it, but i did message you on discord
and we talked about adding a heater via a relay board and integrating it
for control and you told me it wont work,even though all the functionality
is there, you said it couldn't drive it directly, i agreed and said you
would need a transistor to trigger the relay, which would be easily done
via a 3.3v relay board, you then said something about using another esp32
to make it wireless but then you stated other than keeping it functional
you weren't going to work on any new features until you had the large
screen in production as you wanted to make money on that version which i
understand, dev work isn't free and you need to be able to make a living,
but all in all i was simply relaying what you said to me... sorry if my
quick email response wasn't fully clear...