xperiments-in/xtouch #73: Long use of screen disconnects from printer

php4umagento, 2023-12-17:
**Describe the bug**
When screen is used 24/7 on some point it disconnects from printer and shows printer selection list
Have 4 screens and all behave same way

**To Reproduce**
Steps to reproduce the behavior:
1. Leave screen running for over 24hrs
2. Observe printer list

**Expected behavior**
Screen is connected to printer 24/7 unless printer is offline

**XTOUCH (please complete the following information):**
 - FW: 0.9.36

**Printers (please complete the following information):**
 - Device: P1P, P1S
 - FW Version 01.04.02.00

xperiments, 2023-12-18:
Do you have fixed ips?

This is mostly caused because router changed ip

php4umagento, 2023-12-18:
Is there any place where screen shows its MAC or IP?

php4umagento, 2023-12-18:
> Do you have fixed ips?
> 
> This is mostly caused because router changed ip

I will set static and confirm

php4umagento, 2023-12-19:
Static IP fixed it - thanks

closed 2023-12-19 by php4umagento