xperiments-in/xtouch #76: Unable to build using PlatformIO IDE (VSCode)

MaximeLussier, 2023-12-19:
**Describe the bug**
A clear and concise description of what the bug is.
I am unable to build with a clean checkout.

**To Reproduce**
Steps to reproduce the behavior:
1. Clean Checkout
2. Open in VSCode with PlatformIO IDE
3. Build
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
For it to build...

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**XTOUCH (please complete the following information):**
 - Checkout of MAIN

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version MAIN
-  AMS units: 1

**Additional context**
Add any other context about the problem here.
Thanks for the awesome project! Trying to port on SC01 Plus, having it working on the recommended hw presently (web installer)
If I get it working on both boards (no regression...) I will contribute back

xperiments, 2023-12-19:
I am sorry but I don't provide support for compile as actually is a lot tied on my computer. Feel free to enter discord and ask in the os-code-help channel https://discord.com/channels/1144016315548438600/1158808917367865344

closed 2023-12-19 by xperiments

MaximeLussier, 2023-12-19:
Ok, thank! I managed to build but the lv_conf.h file should probably be managed better using LV_CONF_PATH or LV_CONF_INCLUDE_SIMPLE

MaximeLussier, 2023-12-19:
I've managed to port XTouch to the SC01 Plus (3.5" Capacitive). There are a few glitches to correct but it works.
The firmware bug with the P1S is preventing 2 XTouch from working at the same time :(

MaximeLussier, 2023-12-19:
@xperiments Je semble pas avoir accès au Discord...

xperiments, 2023-12-19:
https://discord.gg/Jdvp5CShnt