xperiments-in/xtouch #78: Screen Doesn't Update Current Temperatures or Print Time

Mr-Savag3, 2023-12-22:
**Describe the bug**
The xTouch screen does not update temperature at any time, nor does it show the estimated remaining time when printing.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on your P1P and wait for it to finish booting.
2. Either change the temperature via the P1P's build-in screen, change it via the slicer, or change it by sending a print to the printer via the slicer.
3. Observe that the temperature shown by the xTouch screen does not change.

**Expected behavior**
The xTouch screen should display the current temperature information provided by the printer.

**Screenshots**
[image]

**XTOUCH (please complete the following information):**
 - FW: 0.9.36
 - Connected to the printer itself for power.

**Printers (please complete the following information):**
 - Device: P1P
 - FW Version 01.04.02.00
-  No AMS

**Other Notes**
I'll be updating the firmware of the xTouch once this print is complete to see if that fixes the problem, either by design or by coincidence.

xperiments, 2023-12-23:
Are you in lan mode only?

Mr-Savag3, 2023-12-23:
> Are you in lan mode only?

I am, but it used to work fine and now it doesn't. I haven't changed anything

xperiments, 2023-12-24:
Is a known 1.04 FW bug, You need to work in cloud until Bambu fixes it

closed 2023-12-24 by xperiments

Mr-Savag3, 2023-12-24:
> Is a known 1.04 FW bug, You need to work in cloud until Bambu fixes it

Weird, since it's been working fine since I first set it up. It's also just fixed itself but now it's stopped the slicer from working.... guessing it's another LAN mode issue. If the xTouch is on, OrcaSlicer can't connect to the printer. I'll just deal with it though, it's still better than sending all my data to BBL. 
Thanks for the help!