xperiments-in/xtouch #82: Screen goes to sleep during printing

electricJP, 2023-12-30:
**Describe the bug**
I guess after last update, screen goes to sleep during printing

**To Reproduce**
Steps to reproduce the behavior:
Print, wait 10 min, screen is off

**Expected behavior**
Screen should should be on during printing (Wake on print is On)

**XTOUCH (please complete the following information):**
 - FW: 0.9.58

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version: 01.04.02.00
-  AMS units: 1

flightoffancy, 2024-01-11:
Just added xtouch to my new P1S last night and can confirm that the screen goes to sleep during printing even with "wake on print" toggled on.

mturley777, 2024-01-11:
Not only does mine go to sleep, which I don't mind either way, but it is then dead and will not wake up until the next power cycle.

heelsmj, 2024-01-27:
Wake on print will just turn the screen on if sleeping when the print starts. If you want the screen to remain on while printing disable the timeout function and it will remain on.

FDFantasy, 2024-07-16:
Agree with the above issue, I also have the Wake On Print option selected which I believe was the option needed in the past to keep the screen on during the printing.