xperiments-in/xtouch #83: Xtouch rebooting on connection to printer attempt

hxc0de, 2023-12-30:
**Describe the bug**
Xtouch reboots on attempt to connect to printer. Boots fine, connects to network fine, discovers printer fine. On press of green button to connect to the only printer on the network, it immediately reboots. I have tried 3 different networks. The printer and xtouch are on the same network.

**To Reproduce**
Steps to reproduce the behavior:
1. Boot Printer/Xtouch
2. Observe Xtouch connecting to wifi and discovering printer
3. Click connect button
4. Xtouch reboots

**Expected behavior**
Xtouch should connect fine to the printer, it has done it once or twice before but I would say it is at about a 5% success rate.

**XTOUCH (please complete the following information):**
 - FW: Not sure, I just installed 2 days ago.

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version 01.04.02
-  AMS units: 1

maelstrom27, 2023-12-31:
Was working perfectly until today. Now, when first switched on, xTouch finds the printer and connects when the green button is pressed. During prints the xTouch device seems to lose connection to the printer and produces exactly the same problem as above. My printer shows up and I press the green button, the screen shows "connecting" then "connected" but then immediately goes back to "searching for printers". Resetting the screen makes no difference. If I switch the printer/xTouch off and on again, it searches for the printer and is able to connect again (but doesn't do so automatically).

xTouch firmware 0.9.37

Device: P1S
FW Version 01.04.02.00
No AMS

shraeder, 2024-01-03:
Same here. I was connected fine for several days and then suddenly disconnected. I flashed the firmware to the latest, and it worked for a bit, then stopped again.

Device: P1S
FW Version 01.04.02
1 AMS

MakeItSlow, 2024-01-03:
+1 same behavior as OP.

Device: P1S
Firmware: 01.04.02.00
1 AMS

Gotta wonder if BL is doing this...

fiteoffurdemonz, 2024-01-04:
Same behavior has been happening to me for the past couple of weeks or so. Adding `timeout` and/or `coldboot` to my .json does not fix the boot loop.

Device: P1P[S]
Firmware: 01.05.00.00

TheSithCat, 2024-01-08:
Same behaviour for me, past few days. Adding Coldboot and timeout to .json file didn't help. Tried fresh start, didn't help, can't find my printer.

Device: P1S
Firmware: 01.05.00.00
1 AMS

Vilunki15, 2024-01-09:
Same here. Xtouch is now unusable for me :(

roninniagara, 2024-01-09:
Woke up today. Printer on 1.05.0. Fired up a print - xtouch fails to connect. It sees it, but just reboots over and over.

They do have a new firmware, but I haven't updated to that yet incase.

TCotant, 2024-01-12:
Exactly the same behavior. Reboot cycle "forever" and power cycling both doesn't seem to fix. Was working yesterday prior to the firmware update. Could this be the issue? Just purchased the hardware and printed the case - would be extremely useful to actually use this. Please update with an ETA.

Elessar-lad, 2024-01-14:
I'm facing the same problem as described above. At the "pair printer" screen after I tap the green checkmark the xtouch reboots and starts the finding and pairing process all over again. I tried flashing the firmware again and same results. 

Sadly, this is unusable. I'm reverting the hardware to the original P1P screen.

bpsmicro, 2024-01-14:
I'm on the latest P1S firmware (1.05.1). I didn't have auto updates for xTouch on, but as soon as I manually updated to latest, it died. As a test I deleted all of the json files from the xTouch folder on the microSD. On next boot it requested the access ID again, as expected, but then reverted back to the boot loop failure.
There's likely some incompatibility with the latest P1 firmware and the latest xTouch software. 

I wonder if there are logs generated anywhere? I'll yoink the microSD again tomorrow and go looking.

maelstrom27, 2024-01-14:
I've now updated to the latest P1S firmware and the latest Xtouch firmware and it seems to ne working for me - I'm not sure why. I wonder if there's some setting in one (or both) of the firmwares which causes the problem as I certainly haven't had it for the last few prints I've done.

Elessar-lad, 2024-01-15:
It looks like it's only affecting the P1S. I wonder what's so different from the P1P.

fiteoffurdemonz, 2024-01-15:
> It looks like it's only affecting the P1S. I wonder what's so different from the P1P.

Affecting me on my P1P. I do have the upgrade kit so not sure if that matters.

shraeder, 2024-01-15:
Updated the firmware on both the printer and the dev board, still not working. I've even added the coldboot and timeout lines to the wifi config file. any update on this would be great. I haven't been able to use this for about a month now

shraeder, 2024-01-15:
Removing the xtouch folder from the root of the SD card and reflashing the firmware seems to have resolved this for me.

MakeItSlow, 2024-01-15:
Cleaned the SD card, installed xtouch fresh again and all seem to be working again.

bpsmicro, 2024-01-15:
I ended up doing the same (which was harder than it needed to be because 3 of 4 micro-USB cables I have didn't want to work with it). It worked perfectly ... three times. On the 4th reboot it was back misbehaving again. Now, it will connect to the printer one in maybe 5 reboots. Most of the time it'll keep doing the same as before.
I added the 'coldboot' option too, but I'm not seeing that having any obvious effect other than it takes that much longer for the "list" of printers to come up. Some documentation on what that's supposed to do might be helpful.

pifiuk, 2024-02-10:
Same problem here. Cant connect to my P1P (Upgraded to P1S with kit) or with my P1S, as soon as I validate the printer choice it reboots.

bpsmicro, 2024-02-10:
One thing I discovered is that previously I had the Home Assistant plug-in running. If I had that HA in "LAN" mode and the xTouch in Cloud mode for the MQTT connection, both would work. After a recent update (not 100% sure to which), they won't play together no matter what I do. If the HA module is running, the xTouch will not boot. So something changed in the MQTT stuff that's making them not play nice together.
With Bambu's recent threats to change the firmware to break all these MQTT-based add-ons, this may get worse before it gets better.

simpystance, 2024-04-28:
Same issue here. Is ther a solution for this?

simpystance, 2024-04-28:
Ok fixed mine.
I had a second one that was working, so copyed the sd, and reconnected the printer. All is good now.

closed 2024-06-05 by xperiments