xperiments-in/xtouch #87: Request: If you only have 1 printer, can we setup options on SD card to auto-connect to it

IvanRaide, 2024-01-02:
## Feature Description

Setup a way to auto connect to a specific printer, (useful if you only have 1 printer), or maybe if the list of printers only has 1 item to auto-connect to it?

## Justification

Right now, whenever the xtouch reboots, it shows a list of printers (but there is only one entry) , so invariably whenever I go to the printer the xtouch is almost always showing a single entry to connect to.

## Suggested Implementation (if you have one)

Either settings of the printer ID on the SD card, OR, if the printer list only has 1 entry, auto pick it in say 5 seconds?

## Screenshots or Examples (if applicable)

[Attach screenshots or examples if relevant]

braincrush, 2024-01-02:
I've had this issue when my printer was connected to different router then the xtouch, I've set up a separate 2,4Ghz router and connected both printer and xtouch and now it starts immediately after switching on the printer and goes straight to home screen. I've had all sort of problems like reboots before i set up that router.

IvanRaide, 2024-01-02:
My wifi is controlled by Access Points, (5 of them), and they are in a mesh.  That said, I will take the basement AP (where the printer is) and make a new network/SSID, and bind the xTouch and P1P to it and see if that helps?

braincrush, 2024-01-02:
Worth a try, I almost gave up on xtouch and it was the networks fault, I think there was a mention of issues with mesh networks as well in the documentation so try it out as it works flawlessly for me. Now all I need is that S3 support and it'll turn my P1S into X1 :D

braincrush, 2024-01-02:
Worth a try, I almost gave up on xtouch and it was the networks fault, I think there was a mention of issues with mesh networks as well in the documentations so try it out as it works flawlessly. Now all i need is that S3 support and it'll turn my P1S into X1 :D

IvanRaide, 2024-01-31:
I wanted to add that this seems to have worked.  It's been almost a month now, and by forcing both devices onto one AP (via unique SSID to that ap) of a mesh network, the xTouch is always connected to the P1P.