xperiments-in/xtouch #91: Integrate chamber temperature sensor with Home Assistant using ESPHome

heffe2001, 2024-01-12:
## Feature Description
Integrate the chamber temperature sensor into Home Assistant, possibly using ESPHome?

## Justification
This would allow you to use the chamber temp in automations, track the temperature trends, etc.

## Suggested Implementation (if you have one)

ESPHome in Home Assistant should be able to be used to integrate the sensor into HA.

## Screenshots or Examples (if applicable)

spusuf, 2024-06-27:
A home assistant integration should not rely on connecting to an ESP which then relays to the printer. There is already a community integration. Have a look at the HACS integration here: https://github.com/greghesp/ha-bambulab

Works great for my P1S, I get temp sensors + control, camera viewing, lamp and fan control, print status, etc.

[image]

xperiments, 2024-06-28:
@spusuf What him is asking is to send someway this temp value as is not native in the P1P/P1S and something custom to xtouch

spusuf, 2024-06-28:
> @spusuf What him is asking is to send someway this temp value (as is not native in the P1P/P1S and something custom to xtouch)
> 
> Cheers

As you can see in my screenshot I am using a P1S and there is a temperature sensor in Home Assistant. Nothing custom needed, just works with the Bambu integration.

heffe2001, 2024-06-29:
You're NOT getting a CHAMBER temperature, just a head and bed temp.  I'd like to see the chamber temp like the X1C has on the HA screen.

policeman0077, 2025-04-06:
even better if it has some gpio to control the chamber heater depending on chamber temp reading😉

kmitrop, 2025-04-24:
Any updates regarding this request ?

lecoyote418, 2026-08-21:
For anyone still looking at this feature, I vibe coded it in [my fork](https://github.com/lecoyote418/xtouch). It uses MQTT to transmit the chamber temperature to a MQTT device sensor in Home Assistant.