xperiments-in/xtouch #93: Instead of keeping a value it changes to zero after hitting enter.

bcarriveau, 2024-01-13:
Found when selecting nozzle or bed temps , if you dont change anything but hit enter, it will default to zero instead of keeping the current value. 

At first I didnt realise it happen, but got it heating back up before anything bad happened lol. 

Maybe it can give a confirmation question apon a value change before applying? Then one could say cancel or confirm? 

https://github.com/xperiments-in/xtouch/assets/60801063/d5495889-4568-48d4-98db-43b965aea328

heelsmj, 2024-01-27:
I second this one. This actually messed up a couple prints for me before I realized what was happening because my nozzle temp went to zero while printing.

xperiments, 2024-06-05:
Will check