xperiments-in/xtouch #96: chamber temp sensor is deactivated after re-connect

CameraRick, 2024-01-24:
**Describe the bug**
My xtouch loses connection to the P1S sometimes; this is no big deal, it usually finds the printer again fast and I can just hit the checkmark-button to reconnect. However, anytime this happens, the chamber-temp-sensor is deactivated. I manually have to go into the menu and re-check it. 
I tried restarting xtouch through the menu (the command underneath the chamber temp switch) to "save" my settings, and while it is still enabled after the restart it isn't upon re-connecting after a connection loss.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable the chamber temp sensor in the menu
2. Lose connection between printer and xtouch (probably hard to force, haha) 
3. Reconnect the xtouch to the printer
4. Chamber temp sensor is disabled

**Expected behavior**
I would think this switch should be still enabled after a reconnect

**XTOUCH (please complete the following information):**
 - FW: v0.9.58

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version 01.05.02.00
-  AMS units: 1x (FW 00.00.06.40)

Twan68, 2024-05-09:
Did you solve this problem, and if so, how?

CameraRick, 2024-05-10:
Unfortunately not, I also wouldn't know how. But it doesn't disconnect as often and after a power cycle of the machine, it's there as well

Twan68, 2024-05-12:
I think I found it. 
I Manually set one of the values to "1"instead of "0". It is one of the json files on the card. Is says in the file "chamber temp". 
In one file it was on true. Levae that one. But the other file had the value "0".
I changed it, and it works. 
I also set the latest firmware on the sd-card (which was installed already, but just to be sure). 
I think ifs you replicate this you also might have succes.

xperiments, 2024-06-14:
Update to lattest version and all of this will be gone... cheers

closed 2024-06-14 by xperiments