xperiments-in/xtouch #98: modify minimal sleep time value. Now it is 5 minutes

haadamgit, 2024-01-24:
## Feature Description
The sleep time setting on a slider is 5 minutes minimum. Minimum of 1 minute would be better.

## Justification
Power saving in order to let the device be battery operated.

## Suggested Implementation (if you have one)
Modify minimal slider value to 1 minute

heelsmj, 2024-01-27:
Yes this would be nice that have!

Misterke, 2024-02-08:
I actually have done this in my own local version.  The changes are easy:
- in src/xtouch/types.h change the define of XTOUCH_LCD_MIN_SLEEP_TIME to 1 instead of 5
- in src/ui/components/ui_comp_settingscomponent.c line 265 `lv_slider_set_range(ui_settingsTFTOFFSlider, 4, 60);` change to 4 to a 0

That will still allow value 0 to indicate that no timeout will be applied, but will allow values from 1 up to indicate the number of minutes for the timeout.

In short: it should be a breeze for the author of XTouch to fix this, but since only he can publish new versions to the install and OTA locations, there's little use for me publishing my modifications and I feel to uncertain about some of my changes to risk giving others problems due to them ...

xperiments, 2024-06-14:
@Misterke The actual reason for the 5 is that going to 0 will disable the current feature of going to 0 to completely disable it as some users asked

closed 2024-06-14 by xperiments