xperiments-in/xtouch #100: Feature Request - Locate and print files on printer

Darthmonkey, 2024-01-26:
## Feature Description

The ability to find and print .GCODE from the printer SD card.

## Justification

This feature is already available on the existing LCD screen on P1P and P1S, and is the only thing missing from the xtouch that would allow it to replace the LCD screen.

## Suggested Implementation (if you have one)

It's possible to browse P1P and P1S files via FTP, so if that feature is possible to implement, listing any GCODE in the root of the SD card by date would be possible, and perhaps the option to (re)print it.

RimanCz, 2024-06-24:
It would be awesome!
This is the only reason I didn't remove the original display

xperiments, 2024-06-24:
I18n will come in the future, as 5inch already has it.
Filebrowser will not come to 2.8 I am sorry

closed 2024-06-24 by xperiments