xperiments-in/xtouch #101: change background on the main screen

Buko84, 2024-01-30:
## Feature Description

it would be nice if we could change that xtouch logo to something else, like the printer we have

## Justification

that would personalize the screen to specific printer

Misterke, 2024-02-07:
If I understand the code correctly, that logo is actually the character "4" in the custom font src\ui\fonts\ui_font_xperiments.c and is just rendered as a label with that single character in it using that font.  Not the easiest thing to then make configurable.  I myself don't have much experience with lvgl and have no idea how easy or difficult it would be to read an image from sdcard and display that instead ...

xperiments, 2024-06-05:
Does not adds any functionality.... Maybe future....