xperiments-in/xtouch #104: Question about the chamber temp sensor

PaulBiod, 2024-02-04:
Hi,
I'm about to receive my p1s and preparing an order to buy all the stuff.
About the temp  sensor, if I understood correctly, the sensor will be plugged to the x touch screen. That means the screen has to be close to the printer. But I was thinking about using the screen from another room, so my question is : is it possible to use the sensor plugged to an esp32 so that it s not plugged to the screen, and have xtouch accessing it anyway?
Thanks

Misterke, 2024-02-05:
Obviously that would require changes to the code to fetch the information via WIFI or ESPNOW from the other ESP32.  Not impossible, but I guess there aren't many people in that same situation (especially the author of xtouch), so your best bed would probably be to try to implement it yourself.

PaulBiod, 2024-02-06:
> Obviously that would require changes to the code to fetch the information via WIFI or ESPNOW from the other ESP32. Not impossible, but I guess there aren't many people in that same situation (especially the author of xtouch), so your best bed would probably be to try to implement it yourself.

Thanks @Misterke for the explanation

Tscherno, 2024-02-06:
I would second the feature. I have a Zigbee sensor which is connected via Zigbee2MQTT. So i like the idea to integrate any MQTT value to xtouch. I guess this would be useful to more people.

xperiments, 2024-06-05:
I have already developed a wireless sensor. I will begin selling it in some weeks

closed 2024-06-05 by xperiments