xperiments-in/xtouch #105: Reconnect to last connected printer after IP change / lost connection

php4umagento, 2024-02-08:
## Feature Description

Once a day my router sends new IP to all devices and that causes all Xtouch devices to come back to printer selection screen

## Justification

When this happens I need to select correct printer on all screens with Xtouch which is laborous and error prone.

## Suggested Implementation (if you have one)

When xtouch looses connection with WLAN it should try to select last connected printer rather than showing list of all printers in the network

php4umagento, 2024-02-08:
> I would rather set a fixed IP for your devices or through DHCP reservation.

I have fixed IP and 24 hrs lease period

Misterke, 2024-02-08:
This should not be needed.  DHCP clients normally request extension of their DHCP lease long before it expires and could be assigned to another device.  And since that DHCP address is in use by the device, there's also no reason the server would not grant the extension unless there's a shortage of DHCP addresses in your network.  Bottom-line: getting renewal of the DHCP address should not cause the problem you are stating.  It sounds as if you have some other problem instead (ex. your printer changing IP address or going offline?).  And as mcasteels said: if your printer is indeed getting a different IP address, then configure your DHCP server so that it always does hand out the same IP address for your devices (based on their MAC addresses).

Warsenius, 2024-04-05:
even with the printer having the same IP all the time. every time i start up the xtouch i need to select my printer instead of it automaticly connecting (only have 1 printer so why do i have to press the checkmark)

xperiments, 2024-06-05:
New bet version solves this problems

closed 2024-06-05 by xperiments