xperiments-in/xtouch #106: Chamber Temp wiring is incorrect.

crazyOC8, 2024-02-09:
**Describe the bug**
I bought chamber sensor B and it arrived this morning. I printed the case beforehand then realized the case didn't fit. I wired it up anyway according to the guide and tried turned on the chamber sensor in settings (with screen off my printer). It read "5" then a second later read "-127" so I unplugged it. I grabbed the wires and they burnt me through the rubber. stupid wires. The chamber sensor also does not look like the one that goes in the case for chamber sensor B. 

Steps to reproduce the behavior:
1. Use chamber sensor B and wire it according to the guide.
2. Turn on xtouch.
3. Enable Chamber temp. 
4. Touch wires or pcb and get burnt (do not do this).

**Expected behavior**
I expected to have a working chamber sensor, but it is wired wrong.

**Screenshots**
There are no screenshots. 

**XTOUCH
 - FW: v0.9.58

**Printers
 Printer info is unnecessary, it was not a problem resulting from the printer.

**Additional context**
I think that the cable i was using to wire it may have been phony, but it came with the screen so i think it's right.

bpsmicro, 2024-02-10:
Can you attach photos showing the wires to the ESP32 board, and the opposite ends where they connect to the probe? And the probe itself? Maybe one of us can spot what's awry.

corvankruchten, 2024-04-11:
I had the same issue. Wired as on picture, but -127 and 5.....

xperiments, 2024-06-05:
Any updates?. Seems a bad sensor... I close ATM. Reopen if needed.THX

closed 2024-06-05 by xperiments