xperiments-in/xtouch #108: Missing wifi.json on Display

Narizz28, 2024-03-10:
Since I haven't seen this reported by anyone, I'll add it. I'm getting an error screen "Missing wifi.json" and console "[XTouch][SD] Error Reading JSON File: /wifi.json". I am about to try a new SD card, but in the mean time:

**To Reproduce**
Steps to reproduce the behavior:
1. Install XTouch from https://xperiments.in/xtouch-bin/
2. Reboot
3. Errors noted above

**Expected behavior**
I expected the XTouch to boot to the first screen.

**Screenshots**
[image]

**XTOUCH (please complete the following information):**
 - "version": "0.9.57"

Misterke, 2024-03-12:
Obvious question: did you prepare the SD card as mentioned in the installation instructions?  Is the card "small" enough (i.e. not having too many GBs of storage)?  Did you format it as plain FAT (or ExFAT) using something like [SD card formatter](https://www.sdcard.org/downloads/formatter/)?

Misterke, 2024-03-12:
@Narizz28, I honestly cannot say whether that card will work or not.  I'd suggest you first try formatting it with the SD card formatter, put the appropriate files on it again and see whether that helps.  I'm now also thinking line-endings might play a role, so perhaps best ensure your files use line-feed as end-of-line character (so unix style, not Windows and definitely not MAC).

Narizz28, 2024-03-12:
I double checked line ends first before transferring the file and they were all LF's, not CRLF's. I formatted twice FAT32 and put the wifi.json on after each time.

closed 2024-06-05 by xperiments