xperiments-in/xtouch #110: Black Screen after flash / Rebooting in loop

essafada, 2024-03-13:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**XTOUCH (please complete the following information):**
 - FW: [e.g. 1.0.0]

**Printers (please complete the following information):**
 - Device: [e.g. P1S]
 - FW Version [e.g. xx.xx.xx]
-  AMS units:[e.g. 1]

**Additional context**
Add any other context about the problem here.

essafada, 2024-03-15:
halp plss 😇

dragonnice01, 2024-03-19:
Same problem for me. Mine is a ESP32-2432S024 and not 28R. Can this cause the problem?

essafada, 2024-03-30:
😭

avilshadow, 2024-04-28:
same with 2432S024

avilshadow, 2024-04-28:
> I followed all the instructions carefully, but cannot get the board to turn on or get out off the looping error(see logs)
how did you get logs?

dragonnice01, 2024-04-28:
The 024 is the problem. I tried it with a 28r and it is running perfectly with the same SD card. The screen size is different and most likely this is the issue.

Sascha27, 2024-06-05:
Mein Bild bleibt auch schwarz trotz richtigem Bildschirm, LOG zeigt ständige Reboots an

closed 2024-06-05 by xperiments