xperiments-in/xtouch #118: Chamber Option Gets Disabled After Sleep

3DStyle, 2024-05-18:
**Describe the bug**
I have enabled the chamber option in my xTouch. If I leave my P1S running for a long time of inactivity and come back,
the xTouch returns from sleep, and then I can't see the chamber tempature in the main screen. When I go to the settings
page, I can see that for some reason, the chamber options is disabled. 

**To Reproduce**
Steps to reproduce the behavior:
1. Leave the printer inactivity for few hours and let the xTouch go to sleep.
2. Come back and see if the chamber option is enabled. 
3. It usually happens when I have a print overnight, and come back to my P1S and I can see that chamber is disabled.

**Expected behavior**
xTouch should not disable the chamber option...

**Screenshots**
Not applicable.

**XTOUCH (please complete the following information):**
 - FW: Latest, **v0.9.58**.

**Printers (please complete the following information):**
 - Device: **P1S**
 - FW Version: **01.05.00.00**
-  AMS units: **1**

**Additional context**
Please fix this issue :)

oleyek, 2024-06-07:
The same problem here. But I think that the project has been abandoned... :(

xperiments, 2024-06-14:
Update to the latest version .102 and check if now is ok

@oleyek NOT is not abandoned

closed 2024-06-14 by xperiments

oleyek, 2024-06-16:
Updated to .104 and chamber temp is always visible. But I have another issue. Temperature shows 5 degrees and nothing change after ther firmware upgrade. Sensor is functional, because before upgrade and reboot shows correct temp.

3DStyle, 2024-06-16:
> Updated to .104 and chamber temp is always visible. But I have another issue. Temperature shows 5 degrees and nothing change after ther firmware upgrade. Sensor is functional, because before upgrade and reboot shows correct temp.

Same for me.