xperiments-in/xtouch #119: xtouch no longer functions after 01.05.04.07 Beta firmware update

geekgirlkatie, 2024-05-24:
XTouch will connect to the printer but none of the functions will activate.  This is evidently a known issue caused by a change by Bambu Lab. "Prevent conflicts with Bambu Cloud service by prohibiting printer control through the local MQTT Broker while the printer is logged into Bambu Cloud service."

**XTOUCH (please complete the following information):**
 - FW: 0.9.58

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version 01.05.04.07 Beta
-  AMS units: 1

**Additional context**
The makers of the Panda Touch are having the same issue but evidently going to be releasing a firmware update to resolve the issue. Here's a link to their GitHub page discussing it. https://github.com/bigtreetech/PandaTouch/issues/81

Jer406, 2024-05-25:
having the same issue here.

KevinFairgrieve, 2024-05-26:
It was always going to break. Bambu are making it harder for 3rd party devices to integrate.

My P1S works fine on the current firmware. I may upgrade once xtouch works with it.

shraeder, 2024-05-27:
same problem

1-hexene, 2024-05-29:
### Cause / 原因
Both xtouch and pandatouch works by sending MQTT commands to the machine. In the old versions of printers' firmwares, MQTT command can be sent through LAN and Bambulab's cloud service.
However, BambuLab is now ristricting MQTT commands via LAN when connected to its cloud service. 
It seems they are doing the similar thing as Nintendo did. If you use self-defined devices, you cannot use the cloud service.

xtouch 和 Panda Touch 都通过向打印机发送MQTT指令工作。在旧版本固件里，我们既可以通过局域网发送MQTT指令，也可以通过拓竹云服务发送MQTT指令。
然而，拓竹在新版本固件里做出了限制，当打印机连接到云服务的时候，就不能通过局域网发送MQTT指令。
似乎拓竹正在做和任天堂做相似的事。如果你自定义你的设备，那就别用云服务。
#### Bambulab's update log / 拓竹的更新日志:
- English version:
[image]
- Chinese verion / 中文版本：
[image]
### Possible Solution / 可能的解决方案
1. 
- The developer of BTT Pandatouch states that they will fix this problem by enabling Cloud Mode MQTT commands in the next version of its firmware. Maybe xtouch can also fix the problem by doing so. You can refer to this link:
[https://github.com/bigtreetech/PandaTouch/issues/81#issuecomment-2118865640](url)
- PandaTouch的开发者说，他们会在下个版本的固件了里让Panda Touch能通过云模式发送MQTT指令。xtouch或许也可以通过这个方式解决问题。 你可以参考这个链接：
[https://github.com/bigtreetech/PandaTouch/issues/81#issuecomment-2118865640](url)
2. 
- Sell your Bambulab Printer and buy a Creality K1 series printer.
- 把拓竹卖了换创想三维K1。

shraeder, 2024-05-29:
After digging through the Xtouch discord, it appears that a change that Bambu made to later versions of their firmware. Rolling back to 1.05.02.00 seems to have fixed it for me. I guess we will just have to wait for an update to the xtouch firmware.

If you wan an easy way to downgrade, the bambu handy app makes it easy. Hit the settings icon (three horizontal lines), Firmware version, and then "I want to downgrade to a previous version". select 1.05.02.00 or older and it should work. hope this helps someone else!

xperiments, 2024-05-29:
Fix is done, current testing, patience!!

geekgirlkatie, 2024-05-30:
So Bambu just released a new Beta firmware version 01.05.04.10 and now everything with the xTouch and Panda Touch are working again.

closed 2024-06-05 by xperiments

FAB1150, 2024-11-11:
Sorry to revive this thread, but does "fixed" mean that bambu reverted the changes or that xtouch now connects to bambu's cloud?

I'm thinking of building one for my printer but I'm a bit confused by this change...