xperiments-in/xtouch #124: Impossible to Unload/Load

sfrechette88, 2024-06-08:
**Describe the bug**
I'm able to interract with the print (Light, temps, fan) but I'm unable to start the process to unload and load filament. If I start on the print directly, I can finish the process on the X-Touch.

I just download and install the last firmware today.

My printer is P1P upgrade to P1S

atrambauer, 2024-06-11:
Same problem, updated to last beta firmware and cannot control many things.

pvcHook, 2024-06-11:
Same issue here on all of my 4 printers since updating to the latest firmware.

jdog666, 2024-06-13:
Same issue on all 4 of mine too.

atrambauer, 2024-06-13:
Mine updated today and now it doesn't boot, xtouch screen with Missing config.json

GeoffreySawyer, 2024-06-13:
> Mine updated today and now it doesn't boot, xtouch screen with Missing config.json

Same issue for me too.

xperiments, 2024-06-13:
Please re-read docs and make a new install.

config.json IS new

lvl643charmander, 2024-06-15:
Switching all machines to panda touch now since requiring bambu login credentials.

KevinFairgrieve, 2024-06-15:
> Switching all machines to panda touch now since requiring bambu login credentials. What the actual f*ck is this nonsense.

That is your choice, but Panda touch is not without the same issue.

This from there website!

IMPORTANT NOTE:
The Panda Touch is currently compatible with all Bambu Lab printer firmware versions up until:
- P1P --> v01.05.02.00
- P1S --> v01.05.02.00
- X1C --> v01.07.02.00
- A1 --> v01.01.01.00
- A1 Mini --> v01.01.03.00

It is possible that Bambu Lab may release a future firmware version which impacts the functions available from the Panda Touch. Bigtreetech will monitor all beta firmware versions and if we find a firmware version that affects Panda Touch functionality we will send out an alert via email, Aliexpress chat and our social channels. Buyers can then decide whether or not they would like to perform the update at the risk of losing functionality. Note that the Panda Touch warranty does not cover lost functionality due to a Bambu Lab firmware update being performed.

They will also have to do the same kind of fixes that Xtouch is having to go through. For me I am going to stick to Xtouch and twhen it is released go to the 5" screen.

This is not intended to start a "flame war" it is just my take on things.

xperiments, 2024-06-15:
> Switching all machines to panda touch now since requiring bambu login credentials. What the actual f*ck is this nonsense.

GOGOGO

Remember this project is FREE and the only one F**ink someone is you with your bad words..

Have a good day

PD: We will try to prepare back a version of LAN mode, but we need time and know how BBL manages it... Meanwhile PATIENCE

closed 2024-06-24 by xperiments

xperiments, 2024-07-19:
Do you have any ams connected?

kasuke5, 2024-07-20:
> Do you have any ams connected?

No I don't have ams at all