xperiments-in/xtouch #126: Get "Fatal BBL JWT parsing error"

5Hu, 2024-06-12:
**Describe the bug**
With the latest cloud update I get "Fatal BBL JWT parsing error". After that the diplay restarts.

**To Reproduce**
Steps to reproduce the behavior:
1. OTA Update to latest Display firmware
2. Create config.json with the online tool
3. Save the config.json to sd card
4. reboot display

**Expected behavior**
I expect to see the standard view of the display with temps and menu.

**Screenshots**
Actually not possible, sorry.

**XTOUCH (please complete the following information):**
 - FW: Latest OTA

**Printers (please complete the following information):**
 - Device: P1P with P1S upgrade kit
 - FW Version: 01.06.00.00
-  AMS units: 2

**Additional context**
On start it shows "Connecting -> connected -> Processing JWT Token -> Fatal BBL JWT parsing error".

xperiments, 2024-06-12:
Update to lattest FW 0.9.99 and check another time

closed 2024-06-12 by xperiments

IamanGuava, 2024-06-12:
I'm having the same issue even after updating the firmware to 0.9.99

D1BobbyG, 2024-06-13:
> I'm having the same issue even after updating the firmware to 0.9.99
> 
> Issue solved by disabling 2fa on my Bambu Lab account

How do I disable 2FA please? Any help will be greatly appreciated. Many thanks.

Linsche, 2024-06-13:
If you uses google to sign in, you need to setup a password. See https://xperiments.in/xtouch-bin/config.html at the end of the form. If you have 2FA active its on the same site as the password change.

D1BobbyG, 2024-06-14:
@Linsche thank you very much. Will try that. Had one of those duh moments when I saw 2FA, couldn't think its referring to the 2 step authentication, and to be honest, wouldn't have thought it would affect it. Thanks again!

crossfun, 2024-06-15:
I have the same Problem with the Firmware 0.9.103
2FA is disabled.

Soygen, 2024-06-19:
Simply resetting my password on Bambu's site fixed it for me.

D1BobbyG, 2024-06-24:
> Simply resetting my password on Bambu's site fixed it for me.

Gotta have to try that I guess. It worked for me after turning off 2FA for a couple of days and now it's back on, with 2FA on or off!

D1BobbyG, 2024-06-24:
Sadly, still the same after password reset. Not one to give up on anything, but I think it's worth giving up on this one.

harkness67, 2024-06-24:
Same issue here since this morning : 

> FW updated, 
> Login and PW ok (checked and pasted)
> 2FA OFF

Any solutions ?

NullerDk, 2024-06-24:
Same for me. But worked fine this morning "Fatal BBL JWT parsing error"

MULAWI, 2024-06-25:
Same here... guess if it works again i deactivate OTA... creates only issues!

xperiments, 2024-06-25:
No, OTA will mostly solve issues... because you can get in a too old version

Togroovy, 2024-06-27:
I'm also having this issue.

No 2fa
changed password
updated firmware via sd and OTA

Still stuck in the reboot loop

xperiments, 2024-06-27:
Do an OTA update

MULAWI, 2024-06-27:
Solution is to flash the new version of xtouch. Worked for me

xperiments, 2024-06-27:
> flashed to the new Firmware -> worked 2 Days -> now the same issue again

Buy a X1C

xperiments, 2024-06-27:
I am sorry, TIRED of comments just shitting things...
First I have a live
Second can you think is not a bug produced by me?? Actually Is BBL changing things so blame they
The bug was reported this morning... I have a live... aside this... 
The bug was fixed 1h later I know about it

So if is not ok for you just trow it to the trash and go

hellodosi, 2024-06-27:
Why so rude? If I gave you the feeling that you should sit down immediately to solve the problem, then I'm sorry. I was merely giving an update on how things currently look. Live your life and fix bugs if you feel like it and have the time. but don't shit on users who love your work and give you updates on bugs.

xperiments, 2024-06-27:
I have say sorry before hand...

Rude why? because if you see the timeline, just before you there is a user saying the fix I uploaded is working ok

For me commenting and adding value here is them  check if is really ok and the bug is gone, rather that just confirming the bug  another user has already confirmed is gone

xperiments, 2024-06-27:
Just because I care, and I read all comments here... 
Take time....
Better use this time for dev if not needed here...

Don't take it wrong.... have a nice day....

Delslogos, 2024-06-28:
I have reset the xtouch and after it reinstalled the firmware it had the same issue,  I checked the logs and got this:
Getting accessToken from Bambu Cloud
Received error: 400
deserializeJson() failed: EmptyInput
ets Jul 29 2019 12:21:46
I hope it help you find a fix

xperiments, 2024-06-28:
All seems working on my end

Update to the latest version please and this will be fixed.
If not fixed feel free to come to discord where we can help better

FORMAT SD <- IMPERATIVE
Copy config.json to it
Reinstall from web

Delslogos, 2024-06-28:
So, replaced the cable, formatted sd card, done the erasing and reinstalled the firmware, I also moved the file with the wifi and log in details into the xtouch folder, it then connected and worked... There needs to be an easier way in updating this.

xperiments, 2024-06-28:
Yes and no, OTA is supposed to work, but BBL changes from yesterday left the thing broken...

D1BobbyG, 2024-07-01:
For what is worth guys:|

To start with, thanks PC for creating this thing for us to enjoy.

I been having issues also and the first time, reinstalled the new firmware using the pc, removed 2FA from Bambu account and yes, worked for 2 days and then back to normal.

Tried formatting the card and reinstalling config.json a few times with no luck.

What worked today was installing firmware again, the SD card way, new json on and so far, now it's working again :-).

We just need to keep in mind that gadgets do lose it too every now and then :-).

Thanks again all with constructive feedback.

Triken34, 2024-07-01:
Had the same problem with the fatal error. I did what you said, reinstalled from scratch and redid the file with the .json tool. Fired it up and its running again. I'm running the most recent bbl install.
Thank you for your hard work! This project has made my p1s my favorite printer by far.

Surkaal, 2024-11-04:
Same issue.

I've tried formatting SD card, re-installing the display, disabled 2FA and changed pw.
Out of ideas, any suggestions?

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[XTouch][SCREEN] Setup
[ 676][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x40114124 arg=0x3ffbfdb8
XTouch][SD] SD Card Type: [XTouch][SD] SDHC
[XTouch][SD] SD Card Size: 3781MB
[XTouch][SD] Reading JSON file: /xtouch/settings.json
[XTouch][FS] Checking for touch config
[XTouch][TOUCH] Load from disk
[XTouch][SD] Reading JSON file: /config.json
[XTOUCH][CONNECTION] Connecting to WiFi ..
[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.1.13
[XTouch][SD] Reading JSON file: /config.json
Getting accessToken from Bambu Cloud
deserializeJson() failed: EmptyInput
ets Jul 29 2019 12:21:46