xperiments-in/xtouch #127: New version breaks existing config

inators, 2024-06-12:
**Describe the bug**
I enabled OTA updates because why not.  Well it updated.  And now my existing wifi.conf file is obsolete and you are expecting me to go online and submit my wifi password and bambu lab password to your server for you to do who knows what with to generate the new config.json file.

**Expected behavior**
Tell us how to encode the SSID/Password and Bambu labs credentials onto the new config.json file WITHOUT sending them to you.  I already looked at the javascript on your generation page and I know it is the server creating the config.json file, not the javascript so you are getting our personal data.

kurim, 2024-06-12:
Short dive into source code of the page:
it use javascript btoa to encrypt the value

An ASCII string containing the Base64 representation of stringToEncode.

Also based on the page design, nothing will really send to the server.

inators, 2024-06-12:
OK my Javascript isn't that good.  Appreciate knowing it isn't sent.  Still wish there would have been a conversion or a warning or something instead of just breaking it when it auto updated.

Thanks for your response.

closed 2024-06-12 by inators

kurim, 2024-06-12:
do you know from which version you came, may be it's a one time thing for old version w/o usage of bambu Cloud MQTT

xperiments, 2024-06-13:
@inators Next time we have a discord channel where you can come and all this info is online and updated
Also I updated Readme before update, I have not other ways to manage it...

inators, 2024-06-13:
OK I'm sorry I overreacted a bit.  I'd suggest adding a disclaimer that the config generation is handled entirely client side so dumb people like me don't overreact.

Thanks