xperiments-in/xtouch #129: OTA Update Broke Xtouch Missing config.json

oculardynamic, 2024-06-14:
**Describe the bug**
XTouch updated properly OTA, but upon reboot i get error "Missing config.json"

**To Reproduce**
On boot up

**Expected behavior**
Should boot up

**XTOUCH (please complete the following information):**
Newest Version

**Printers (please complete the following information):**
For P1P

Not sure what to do in order to get it to boot up. Please Advise

oculardynamic, 2024-06-14:
I was able to remedy the issue and will post it here for anyone else who comes across it

Create a Configuration File:

Use the online configuration [TOOL](https://xperiments.in/xtouch-bin/config.html) to generate a config.json file.
Place it on the root directory of the SD card.
Your SD card is now properly initialized with the required configuration. This configuration will enable the xtouch screen to connect to your WiFi network and BambuLab servers during the installation process.

EnergySaversAi, 2024-06-14:
I have the same issue with chamber temperature stuck on 5

oculardynamic, 2024-06-14:
> I have the same issue with chamber temperature stuck on 5

Good to know it was the update that broke the temp sensor. Hopefully we can get that fixed

wi1k1n, 2024-06-14:
I have the same setup, but instead of OTA I flashed firmware using online flashing tool. No issues with DS18B20 🤔

EnergySaversAi, 2024-06-14:
I also flashed by firmware.bin no temp sensor always 5

oculardynamic, 2024-06-14:
I think i read on facebook you have to reformat the sdcard if you do the OTA

xperiments, 2024-06-14:
All of you, can say what printer model. I am unable to reproduce it in P1P

puckshaw, 2024-06-15:
My chamber temp sensor is stuck on 5 also. P1S. OTA update and created new config file.

xperiments, 2024-06-15:
And formatted SD?

xperiments, 2024-06-15:
Could also you post in zip the contents of xtouch folder

xperiments, 2024-06-15:
I am closing it, as I have tested and retested and the only logic here to fail is that you have not formatted the SD card

Please

Format your SD card
Copy config.json to it
Reboot screen
Setup touch etc

Please reopen if them does not work and also come to Discord where is more easy to help in this things

THX!!

xperiments, 2024-06-15:
OK reopened as I have found the bug...

Working on it...

xperiments, 2024-06-15:
Fixed in v0.9.104.

Install. Set chamber on another time
Reboot

closed 2024-06-15 by xperiments

oculardynamic, 2024-06-15:
> Fixed in v0.9.104.
> 
> Install. Set chamber on another time Reboot

OTA Installed, chamber still showing 5, but i'm not exactly what "Set Chamber on another time" means? I still dont see it in my settings. Should i set the chamber sensor to "1" in the settings files as its shows set to "0"

xperiments, 2024-06-15:
Have y formated the SD?

Moe555, 2024-06-22:
I had the same issue.
I can confirm that after formatting SD, copying config.json, reinstalling via USB and setting the chamber temp to on afterwards it's working fine again.