xperiments-in/xtouch #130: return lan mode

bOBERmASTER, 2024-06-15:
i'm and other people DON'T uses cloud mode this mode is not secured. return lan mode please

xperiments, 2024-06-16:
Only way ATM is rebert to old version, but before disable ota updates

them dowmload 

https://github.com/xperiments/xtouch-bin/blob/e5587533ef98b3b54d471a3c7a4880a7bf33de92/fw/firmware.bin

create the old wifi.json

put it all wifi.json and firmware.bin at sd root

delete xtouch folder from sd

reboot

closed 2024-06-16 by xperiments

cwkean87, 2024-06-21:
works like a charm. thank you!

Mr-Savag3, 2024-06-24:
THANK YOU. I was so upset when I realized I was locked to Cloud Mode. It's bad enough that BBL is pushing it so hard.

xperiments, 2024-06-24:
Just wait for new things... as I use cloud for control printer, not for sending print files... they remain local...

MisterTee8227, 2025-01-26:
Is anyone working on adding official support for Lan Only mode in the newest version of the X-Touch firmware?  Seems like it should be easy to do since they have the code from the older version.  This is the only thing preventing me from going LAN only.

Craftzman7, 2025-02-10:
I got LAN mode working on the latest firmware with some code modifications, if enough people want it I'll publish it. The modifications are pretty trivial though.

Craftzman7, 2025-02-10:
I have published my LAN firmware, complete with new install instructions. https://github.com/Craftzman7/xtouch

travisterrell, 2025-04-06:
Why was this closed? LAN mode still isn't present, and the other tickets were closed as duplicates of this one. 

I see a comment elsewhere from the main developer that says maintaining 2 different branches is too much work, since Bambu is likely to do some changes soon and introduce developer mode. That makes sense! _However,_ without an open ticket, there's no way to track progress on it. (Unless this project is discontinuing LAN Mode permanently.)

Thanks