xperiments-in/xtouch #131: Ability to Interface with Printer Broken w/ Bambu Labs update 1.6.??

VPrintsUS, 2024-06-15:
Haven't seen this here yet so starting the thread despite never using Github before. Upon updating my P1S to 1.6.?? I had to reupload the config.json file to my XTouch SD card. After connecting to the printer, ie the XTouch interface appears connected, pressing any of the buttons on the XTouch does not cause the printer to react.

Never had this issue until today 06/15. I believe the update for the printers came out a few days ago?

Sorry if I miss important details. Goodluck and hope this can be resolved!

Jer406, 2024-06-16:
same here too

xperiments, 2024-06-16:
Have you formatted the SD card? its IMPERATIVE

Nosler1975, 2024-06-16:
I have similar issues, hopefully there is a fix on its way..

Jer406, 2024-06-17:
formatted the sd card. 
used the tool to make the .json
problem is 100% fixed

Nosler1975, 2024-06-22:
Tried repeatedly to do the Config set up but no good, when i open the file created by the config, it does not have any of my correct details.

I am using a gmail account.

xperiments, 2024-06-22:
@Nosler1975 The file is ok aside you don't see your info
If using gmail check to reset password of BBL
If done disbale 2FA

bdasshare, 2024-06-22:
Same problem, guess Bambu has broken there API or there must be some new method.

Result i get after debugging was an empty response on the accesToken in the json response and also with Postman.
{
    "accessToken": "",
    "refreshToken": "",
    "expiresIn": 0,
    "refreshExpiresIn": 0,
    "tfaKey": "xxxxx",
    "accessMethod": ""
}

closed 2024-06-22 by xperiments