xperiments-in/xtouch #132: JWT parsing error

Loadie150, 2024-06-15:
**Describe the bug**
My P1s has been turned off for 1 week, when I hooked it back up today I read not able to connect. I went in and reinstalled both the .json file and firmware from the Xtouch installer page. When the ESP32 boots up it reads, connecting, then connected, then processing JWT token, then finally FATAL BBl JWT parsing error. This just cycles over and over again.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**XTOUCH (please complete the following information):**
 - FW: [not sure, latest one on github site]

**Printers (please complete the following information):**
 - Device: [ P1S]
 - FW Version [e.g. 01.06.00.00]
-  AMS units: [2]

**Additional context**
Not sure what to do ?

Loadie150, 2024-06-15:
I also formatted the SD card to FAT32 before redoing everything.

xperiments, 2024-06-16:
Have you made the config here ?

https://xperiments.in/xtouch-bin/config.html

Loadie150, 2024-06-16:
Yes I did. I originally did the notepad way, but that didn’t work. So I reformatted the SD card and used the link then the xtouch uploader.

telematique89, 2024-06-16:
I have the Same issue.

xperiments, 2024-06-16:
Have y 2FA enabled, if so disable it

telematique89, 2024-06-16:
It was the 2FA. thanks !

xperiments, 2024-06-16:
Great

crossfun, 2024-06-16:
I have the same Problem with the Firmware 0.9.103
2FA is disabled.

xperiments, 2024-06-16:
And you have entered your bambu password or your gmail password? Needs to be from bambu account

crossfun, 2024-06-16:
I have entered my Bambu Login. GMX Mail and the Bambu Password.

Loadie150, 2024-06-16:
> Have y 2FA enabled, if so disable it

What’s 2FA?

Loadie150, 2024-06-16:
Problem resolved! Fat finger stupid mistake. Error on password, lol.. Still don't know what 2FA is though.

xperiments, 2024-06-16:
2fa two factor authentication...

So the process of receiving a Code or have An app for that that validates
IS you

crossfun, 2024-06-16:
I double-checked my 2FA disabled twice, formatted the SD card again, and created a new configuration file. The password I entered was copied and pasted. I am still receiving the same error.

xperiments, 2024-06-16:
Only thing I can say is come to Discord where I am mostly avaliable all time and just ask
So we can try to see whats the problem live...
Anyone reported not able to connect after solving password reset and disabling 2FA
BTW you are in China?
Thats the only servers I was unable to test

xperiments, 2024-06-17:
Closed in 0.9.115

A lot of thanks to @crossfun for his test and support!!

closed 2024-06-17 by xperiments