xperiments-in/xtouch #136: Fatal BWL JWT Parsing Error

Mavrap, 2024-06-24:
Describe the bug
With the latest cloud update I get "Fatal BBL JWT parsing error". After that the display restarts.

To Reproduce
Steps to reproduce the behavior:

update with the Firmware.bin 
Create config.json with the online tool
Save the config.json to sd card
reboot display
Expected behavior
I expect to see the standard view of the display with temps and menu.

Screenshots
Actually not possible, sorry.
but the software loops :
Connecting -> connected -> Checking for OTA update -> Processing JWT Token -> Fatal BBL JWT parsing error

XTOUCH (please complete the following information): last firmware

FW: Latest OTA
Printers (please complete the following information):

Device: P1S
FW Version: 01.06.00.00
AMS units: 1
Additional context
On start it shows "Connecting -> connected -> Checking for OTA update -> Processing JWT Token -> Fatal BBL JWT parsing error".
2FA is disabled. I changed my password without any effect (Bambulab account)

Sascha27, 2024-06-24:
I have exactly the same problem since today, same printer, same firmware
Ich habe genau das selbe Problem seit heute, selber Drucker selbe Firmware
-Wifi connected
-Processing JWT Token
-FATAL BBL JWT PARSING ERROR

harkness67, 2024-06-24:
Same here since this morning too

btastic, 2024-06-24:
Manually doing the steps the code does, seems to work fine.

1. Getting the token via the login endpoint still works
2. Decoding the token still includes the username
3. Here is the log from the device:

```
[...]
Getting accessToken from Bambu Cloud
[ 14865][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[ 14879][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -9984
Received error: -1
deserializeJson() failed: EmptyInput
```

So it appers as the ssl certificate is not being validated correctly anymore. Not sure how to fix this.

Alan-Somebody, 2024-06-24:
I have two xTouch devices and both have the same error as of this morning.  I formatted one of their SD cards and re-made the config file, but same error occurs.

btastic, 2024-06-24:
I created a Pull request to fix this issue: #138

Docomar750, 2024-06-24:
Same Issue here worked on a board setting up yesterday but had to send that back as touch screen not responding just setting new one up today and no connecting to Bambu getting "Fatal BBL JWT parsing error!!!!

Docomar750, 2024-06-24:
[XTouch][SCREEN] Setup
[   670][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x40113424 arg=0x3ffbfdc8
XTouch][SD] SD Card Type: [XTouch][SD] SDHC
[XTouch][SD] SD Card Size: 29818MB
[XTouch][SD] Reading JSON file: /xtouch/settings.json
[XTouch][FS] Checking for touch config
[XTouch][TOUCH] Load from disk
[XTouch][SD] Reading JSON file: /config.json
[XTOUCH][CONNECTION] Connecting to WiFi ..
[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.1.70
[XTouch][SD] Reading JSON file: /config.json
Getting accessToken from Bambu Cloud
[  7837][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[  7852][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -9984
Received error: -1
deserializeJson() failed: EmptyInput
ets Jul 29 2019 12:21:46

harkness67, 2024-06-24:
Solved here, thanks for update !

Mavrap, 2024-06-24:
Thanks a lot!!! It works

closed 2024-06-24 by xperiments