xperiments-in/xtouch #137: FATAL BBL JWT PARSING ERROR since today

Sascha27, 2024-06-24:
Since today I have the problem that Xtouch is stuck in a loop.

Even another firmware update with firmware.bin did not solve the problem.

Printer P1S

Firmware 01.06.00.00

Xtouch tries this all the time:
-WiFi connected
-Processing JWT Token
-FATAL BBL JWT PARSING ERROR

harkness67, 2024-06-24:
Same here.. since this morning too.

-FW updated,
-Login and PW ok (checked and pasted)
-2FA OFF

reflex1124, 2024-06-24:
Same.
I have been experiencing this problem for the past few hours.

I have reset the password and reviewed the configuration.
2FA is turned off.

Alan-Somebody, 2024-06-24:
I have two xTouch devices and both have the same error as of this morning.  I formatted one of their SD cards and re-made the config file, but same error occurs.

xperiments, 2024-06-24:
Update to latest version...

closed 2024-06-24 by xperiments

Docomar750, 2024-06-24:
Just reloaded

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[XTouch][SCREEN] Setup
[   671][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x4011400c arg=0x3ffbfdc8
XTouch][SD] SD Card Type: [XTouch][SD] SDHC
[XTouch][SD] SD Card Size: 29818MB
[XTouch][SD] Reading JSON file: /xtouch/settings.json
[XTouch][FS] Checking for touch config
[XTouch][TOUCH] Load from disk
[XTouch][SD] Reading JSON file: /config.json
[XTOUCH][CONNECTION] Connecting to WiFi ..
[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.1.70
[XTouch][SD] Reading JSON file: /config.json
Getting accessToken from Bambu Cloud
Received error: 400
deserializeJson() failed: EmptyInput
ets Jul 29 2019 12:21:46

Surkaal, 2024-11-04:
Same issue. 

I've tried formatting SD card, re-installing the display, disabled 2FA and changed pw. 
Out of ideas, any suggestions?

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[XTouch][SCREEN] Setup
[   676][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x40114124 arg=0x3ffbfdb8
XTouch][SD] SD Card Type: [XTouch][SD] SDHC
[XTouch][SD] SD Card Size: 3781MB
[XTouch][SD] Reading JSON file: /xtouch/settings.json
[XTouch][FS] Checking for touch config
[XTouch][TOUCH] Load from disk
[XTouch][SD] Reading JSON file: /config.json
[XTOUCH][CONNECTION] Connecting to WiFi ..
[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.1.13
[XTouch][SD] Reading JSON file: /config.json
Getting accessToken from Bambu Cloud
deserializeJson() failed: EmptyInput
ets Jul 29 2019 12:21:46