xperiments-in/xtouch #140: Fatal BBLJWT parsing error since Jun 24, 2024

Rubiksensei, 2024-06-26:
**Describe the bug**
Boot loop with a "Fatal BBLJWT parsing error" message on screen.

**To Reproduce**
Just power on the device.

**Expected behavior**
The device to power on, connect to WIFI, check update, then to log to Bambu account.

**XTOUCH (please complete the following information):**
 - FW: 0.9.132

**Printers:**
 - Device: P1S
 - FW Version: 01.06.00.00 [latest to date]
-  AMS units: 1

**Additional context**
What I tried so far :
- Check WIFI and bambu credentials
- Reset Bambu credentials (with a new config file generated each time)
- Enabled then disabled 2FA
- Log out then log in the printer
- Check update of the xtouch
- Manually update the xtouch
- Clean install of the xtouch following instructions

Log :
`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[XTouch][SCREEN] Setup
[   675][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x4011400c arg=0x3ffbfdc8
XTouch][SD] SD Card Type: [XTouch][SD] SDHC
[XTouch][SD] SD Card Size: 3823MB
[XTouch][SD] Reading JSON file: /xtouch/settings.json
[XTouch][SD] Reading JSON file: /xtouch/printer-pair.json
[XTouch][SD] Reading JSON file: /xtouch/printer-pair.json
[XTouch][SD] Reading JSON file: /xtouch/printer.json
[XTouch][FS] Checking for touch config
[XTouch][TOUCH] Load from disk
[XTouch][SD] Reading JSON file: /config.json
[XTOUCH][CONNECTION] Connecting to WiFi ..
[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.1.53
[XTouch][SD] Reading JSON file: /config.json
Getting accessToken from Bambu Cloud
Received error: 400
deserializeJson() failed: EmptyInput
ets Jul 29 2019 12:21:46`

Rubiksensei, 2024-06-26:
Somehow managed to downgrade to 0.9.131 (not sure how, my computer crashed in the middle/because of the process)
The xtouch came back to life. OTA update is off by default, and I will leave it as it is.
I'm not closing in case there is something to do to solve the issue with 0.9.132

xperiments, 2024-06-26:
I am closing as .131 is the latest version
There is not issue, y just remained in an unknowns status due to a BBL change in the certificates and the only way to solve it is installing from web or installing the lattest FW via SD

closed 2024-06-26 by xperiments

LordKlaus, 2024-06-27:
Installed from web installed several times, formatted SD in FAT32 several times. Lastest firmware installed. Bambulab password changed several times. It works one day and then the day after no longer works.

xperiments, 2024-06-27:
> Installed from web installed several times, formatted SD in FAT32 several times. Lastest firmware installed. Bambulab password changed several times. It works one day and then the day after no longer works.

Not my fault if one day works another not, ask BBL

Reinstall. Fix is online