xperiments-in/xtouch #148: Fatal BBLJWT parsing error

Gandalf581, 2024-07-05:
**Describe the bug**
Boot loop with a "Fatal BBLJWT parsing error" message on screen.

-WiFi connected
-Processing JWT Token
-FATAL BBL JWT PARSING ERROR

**To Reproduce**
Steps to reproduce the behavior:
Power on the device

**Expected behavior**
The device to power on, connect to WIFI and then to log to Bambu account.

**Screenshots**
not available

**XTOUCH (please complete the following information):**
 - FW: 0.9.140

**Printers (please complete the following information):**
 - Device: P1P
 - FW Version: 01.06.00.00
-  AMS units: 0

**Additional context**
- formatted SD card multiple times with FAT32
- did FW update on xTouch (download from: https://xperiments.in/xtouch-bin/fw/firmware.bin)
- created config.json multiple times with Bambulab account (no gmail) from https://xperiments.in/xtouch-bin/config.html
- executed xtouch web installer mutliple times (https://xperiments.in/xtouch-bin/)

xperiments, 2024-07-05:
Really hard to know, please come to Discord where I can support better

xperiments, 2024-07-06:
Closed as resolved

closed 2024-07-06 by xperiments