xperiments-in/xtouch #151: Explain config.json format

dayne, 2024-07-07:
## Feature Description

Describe the format of the config.json so people can make it without using an online form that submits their local wifi details and Bambu login details to a website. 

## Justification

Securing & future proofing: Reduce the need to submit private information to an online service that might be trusted now but could be compromised or become an attack surface (now or in the future). 

## Suggested Implementation (if you have one)

- Explain how the `config.json` fields are generated and how to use them.  
- Offer a locally hostable `config.json` generator that spits out `config.json` w/o submitting to an online service.

xperiments, 2024-07-07:
Hi the website don't send any info anywhere all is done in local, you can check the code.
The format is simple, all keys are Base64 encoded except cloud-region

{
  "ssid": "BASE_64_ENCODED_SSID",
  "pwd": "BASE_64_ENCODED_SSID",
  "cloud-email": "BASE_64_BBL_EMAIL",
  "cloud-password": "BASE_64_BBL_PASSWORD",
  "cloud-region": "World" // World or China
}

closed 2024-07-07 by xperiments

dayne, 2024-07-10:
Thank you for the speed response and the concise and clear explication.  

Great project.