xperiments-in/xtouch #153: Could not connect to region locked P1S using LAN only mode

WilliamLi564, 2024-07-10:
I tried using Global option when making config file and I couldn't connect to my regon locked P1S. And I do not have a China bambu account to use in China region for config file. And during startup I do not have searching for printers page and unlinked printer does not do anything, it just restart Xtouch and go back to my linked A1.

**XTOUCH (please complete the following information):**
 - FW: V0.9.140

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version 01.04.00.00
-  AMS units: 1

xperiments, 2024-07-10:
Please come to discord where I can try to help, but will be hard.
If worked before with the wifi.json file them revert back to that version

closed 2024-07-10 by xperiments

WilliamLi564, 2024-07-11:
Thank you very much. Issues solved!!