xperiments-in/xtouch #157: Touchscreen trouble

lukasz2194, 2024-08-04:
## Feature Description

[Detailed description of the new feature you're requesting]

## Justification

[Reasons why you believe this feature would be useful]

## Suggested Implementation (if you have one)

[Ideas on how the feature could be implemented]

## Screenshots or Examples (if applicable)

[Attach screenshots or examples if relevant]

lukasz2194, 2024-08-04:
After installing, my xtouch does not respond to touch. It displays changes such as turning on the printer's light or changing the nozzle temperature, but I cannot control anything from the screen. The touch screen works and calibration can be performed. What this is about ?

lukasz2194, 2024-08-04:
Solution : Recalibrate touchscreen propertly :P

jazzyboy, 2024-10-07:
I'm in the same case : the install was Ok, the appairing to the printer Ok, the screen display the xtouch desktop, but no interactions is possible with the screen.
The colors seems to be very light and not the same as i could see on the youtube channels (less contrast)
[image]

lukasz2194, 2024-10-08:
Try to reinstall and calibrate touchscreen again after installation

Quanghoster, 2024-11-11:
Seeing the same thing here.  Did anyone found a solution?  Reinstalling didn't resolve

closed 2024-11-11 by xperiments

Quanghoster, 2024-11-11:
> Seeing the same thing here. Did anyone found a solution? Reinstalling didn't resolve

My bad - I was touching the x in the text instead of the one in the top left of the screen when calibrating, lol.  It works perfectly now I'm not being an idiot