xperiments-in/xtouch #161: WiFi password requirements

JohnLinford, 2024-09-07:
I used an earlier version of Xtouch and it worked well. Recently I decided to update to the latest version, which I discover requires a config file, created using the "config.json easy setup" utility. 

Unfortunately this utility requires a WiFi password. None of my WiFi access points have password set, as I live in the middle of nowhere and have no need to make my networks secure. I tried entering a dummy password, thinking that I would simply edit the config.json file manually but then I discovered that it is encrypted! This means I can no longer use Xtouch.

My suggestion is that the password requirements are relaxed. There is no need to impose WiFi password requirements at all. Just allow anything... or nothing. Alternatively, do not encrypt the config.json file: it's only going onto a USB memory stick so I can't see why encryption should be considered necessary at all.

Thank you for considering this request.

John

umeiko, 2024-09-11:
It seems encrypted with base64，maybe you can simply try an online base64 encoder to solve this?

closed 2024-11-11 by xperiments

xperiments, 2024-11-11:
Up to you, you ask me to change the code to just support you, the only person in this world without a wifi password?

I am using 2024 standards if you are not fine just get back to 2000's

Cheers